AllanGallop/mini-web-radio #5: Pinout / DAC Wiring

derFrickler, 2022-08-18:
Just finished my build, but i don't get any sound. Now i figured out there is something strange with the wiring of the CAD module.

The instructions say that the DAC module should be soldered with the GND Pins aligning. 
Which leads to: 
- SD 12
- Gain 14
- DIN 27
- BCLK 26
- LRC 25
-This is although what it looks lite on the pics: https://content.instructables.com/ORIG/FPW/Q9T0/L66H6K5D/FPWQ9T0L66H6K5D.jpg

But the Code defines these Pins as:
#define I2S_DOUT 26
#define I2S_BCLK 25
#define I2S_LRC  33
#define I2S_GAIN 27
#define I2S_SD 14

Which is shifted by one pin. Could you please clarify this.

Thanks a lot
       Johannes

AllanGallop, 2022-08-18:
Hi,
Apologies you are correct, we modified the design for the instructables slightly, moving the i2s module down by one pin which the code and schematic do not reflect yet. An update will be going up by end of the week to correct this and add deepsleep, but until then changing the pin definitions will facilitate getting it working:

SD = 12
Gain = 14
DIN = 27
BCLK = 26
LRC = 25

derFrickler, 2022-08-18:
YES, already got it working by changing the code. 
#define I2S_DOUT 27
#define I2S_BCLK 26
#define I2S_LRC  25
#define I2S_GAIN 14
#define I2S_SD 12

BTW, the defined BATT pin is not used, analogread is hardcoded to pin 35.

Thanks 
     Johannes

AllanGallop, 2022-08-18:
Repo has been updated to [v1.3.0](https://github.com/AllanGallop/mini-web-radio/releases/tag/v1.3.0) which will correct these issues

closed 2022-08-18 by AllanGallop