AllanGallop/mini-web-radio #7: cannot get response from 192.168.1.1

ranger2012, 2023-04-06:
i have logon to MWR-WIFI-SETUP
removed the jumper that connects pin 13 & 15

but  browser complain of 
"no webpage was found  http://192.168.1.1/"
nothing comes out of the said address..

AllanGallop, 2023-04-06:
Hi,
Could you dump the output from the serial console from startup to point you try and connect please. Initially I would double check the data folder has been uploaded as this usually occurs when spiffs is empty

ranger2012, 2023-04-07:
i can see occasional  "inbound request" nothing else is happening..

attached is the serial monitor log..

5:34:06.329 -> RSSI:0
15:34:07.824 -> Battery:6.83
15:34:07.824 -> RSSI:0
15:34:09.345 -> Battery:6.83
15:34:09.345 -> RSSI:0
15:34:10.840 -> Battery:6.83
15:34:10.840 -> RSSI:0
15:34:12.338 -> Battery:6.83
15:34:12.391 -> RSSI:0
15:34:14.402 -> SPIFFS mounted successfully
15:34:14.402 -> 0
15:34:14.402 -> Battery:6.83
15:34:14.402 -> RSSI:0
15:34:14.402 -> Configuration Jumper Detected
15:34:14.402 -> Starting WiFi in AP Mode
15:34:15.877 -> Battery:6.83
15:34:15.877 -> RSSI:0
15:34:17.401 -> Battery:6.83
15:34:17.401 -> RSSI:0
15:34:18.923 -> Battery:6.83
15:34:18.923 -> RSSI:0
15:34:20.424 -> Battery:6.83
15:34:20.424 -> RSSI:0
15:34:21.908 -> Battery:6.83
15:34:21.908 -> RSSI:0
15:34:23.424 -> Battery:6.83
15:34:23.424 -> RSSI:0
15:34:24.926 -> Battery:6.83
15:34:24.926 -> RSSI:0
15:34:26.441 -> Battery:6.83
15:34:26.441 -> RSSI:0
15:34:27.934 -> Battery:6.83
15:34:27.934 -> RSSI:0
15:34:29.456 -> Battery:6.83
15:34:29.456 -> RSSI:0
15:34:30.959 -> Battery:6.83
15:34:30.959 -> RSSI:0
15:34:32.452 -> Battery:6.83
15:34:32.452 -> RSSI:0
15:34:33.945 -> Battery:6.83
15:34:33.945 -> RSSI:0
15:34:35.474 -> Battery:6.83
15:34:35.474 -> RSSI:0
15:34:36.982 -> Battery:6.83
15:34:36.982 -> RSSI:0
15:34:38.475 -> Battery:6.83
15:34:38.475 -> RSSI:0
15:34:40.129 -> Battery:6.83
15:34:40.129 -> RSSI:0
15:34:41.467 -> Battery:6.83
15:34:41.467 -> RSSI:0
15:34:42.976 -> Battery:6.83
15:34:42.976 -> RSSI:0
15:34:44.472 -> Battery:6.83
15:34:44.472 -> RSSI:0
15:34:46.019 -> Battery:6.83
15:34:46.019 -> RSSI:0
15:34:47.502 -> Battery:6.83
15:34:47.502 -> RSSI:0
15:34:49.014 -> Battery:6.83
15:34:49.014 -> RSSI:0
15:34:50.518 -> Battery:6.83
15:34:50.518 -> RSSI:0
15:34:52.026 -> Battery:6.83
15:34:52.026 -> RSSI:0
15:34:53.519 -> Battery:6.83
15:34:53.519 -> RSSI:0
15:34:55.019 -> Battery:6.83
15:34:55.019 -> RSSI:0
15:34:56.513 -> Battery:6.83
15:34:56.513 -> RSSI:0
15:34:58.015 -> Battery:6.83
15:34:58.015 -> RSSI:0
15:34:59.548 -> Battery:6.83
15:34:59.548 -> RSSI:0
15:35:01.024 -> Battery:6.83
15:35:01.024 -> RSSI:0
15:35:02.572 -> Battery:6.83
15:35:02.572 -> RSSI:0
15:35:04.052 -> Battery:6.83
15:35:04.052 -> RSSI:0
15:35:05.537 -> Battery:6.83
15:35:05.537 -> RSSI:0
15:35:07.078 -> Battery:6.83
15:35:07.078 -> RSSI:0
15:35:07.166 -> Inbound Request
15:35:08.577 -> Battery:6.83
15:35:08.577 -> RSSI:0
15:35:10.061 -> Battery:6.83
15:35:10.061 -> RSSI:0
15:35:11.581 -> Battery:6.83
15:35:11.581 -> RSSI:0
15:35:12.467 -> Inbound Request
15:35:13.092 -> Battery:6.83
15:35:13.092 -> RSSI:0
15:35:13.939 -> Inbound Request
15:35:14.609 -> Battery:6.83
15:35:14.609 -> RSSI:0
15:35:16.086 -> Battery:6.83
15:35:16.086 -> RSSI:0
15:35:16.433 -> Inbound Request
15:35:17.606 -> Battery:6.83
15:35:17.606 -> RSSI:0
15:35:17.714 -> Inbound Request
15:35:19.087 -> Inbound Request
15:35:19.087 -> Battery:6.83
15:35:19.087 -> RSSI:0
15:35:20.593 -> Battery:6.83
15:35:20.634 -> RSSI:0
15:35:22.123 -> Battery:6.83
15:35:22.123 -> RSSI:0
15:35:23.608 -> Battery:6.83

AllanGallop, 2023-04-07:
Looks like the data folder hasn't uploaded, have you tried uploading it again via Tools > ESP32 Sketch Data Upload in the Arduino IDE?

ranger2012, 2023-04-07:
i cannot find  arduino/tools/upload   opton.??

Sketch/upload or control U, dont have option to  select data

AllanGallop, 2023-04-07:
It's part of EspTools package, instructions to install are in the Wiki:
https://github.com/AllanGallop/mini-web-radio/wiki/Arduino-IDE-(from-Scratch)

ranger2012, 2023-04-07:
I believe,  miss this step, uploading the  data folder into the spiff,   but how do i do that, there's no option n my arduino 2.04, or in arduino 1.8.19

Thanks,

AllanGallop, 2023-04-07:
I don't believe there is a version of esptools for IDE 2.x yet, but certainly is for 1.8.x

Full instructions can be found on their [GitHub ](https://github.com/me-no-dev/arduino-esp32fs-plugin)

ranger2012, 2023-04-07:
All working now, works like a charm..

Thank you very much

closed 2023-04-07 by ranger2012