AllanGallop/mini-web-radio #8: Powered by micro USB without batteries.

mikronwfh, 2023-08-04:
Hi! All I did was remove line 49 from the script.

1. What can I do next to use micro USB power only. I will never use battery power.
2. Where to connect the potentiometer in this case?
3. Can speakers above 3W be inefficient and will play unstable thanks to the radio?

[image]

AllanGallop, 2023-08-04:
Hi,
I've created a [new branch](https://github.com/AllanGallop/mini-web-radio/tree/no-battery) with a modified firmware INO file and a schematic change (see the readme). This removes the need for the battery LED (R3 and D2) along with omitting the battery sense circuit (R1/R2) and instead connecting the switch between the 3v3 rail and GPIO pin 33 to sleep/wake the ESP32.

AllanGallop, 2023-08-04:
To answer your questions more directly:

1. Use the new branch and modification to remove the battery and it's associated parts
2. The potentiometer remains unchanged however the switch side of it connects directly between 3v3 and Pin 33
3. The maximum output of the I²S module is 3W so if using a 4Ω speaker you want it to have a power rating equal or greater than 3W (>1.5W for an 8Ω speaker). If attempting to drive an exceptionally large speaker the module may have issues actuating the voice coil

closed 2023-08-04 by AllanGallop

mikronwfh, 2023-08-04:
thank you very much.  I'll let you know how it works when I test it :)

mikronwfh, 2023-08-06:
Can I connect the aux socket instead of the speaker to output the sound to the receiver and amplifier?

AllanGallop, 2023-08-06:
You can but bare in mind the output is mono not stereo, if that's not an issue you can bridge the channels to drive a stereo system. If it needs to be line level then you'd need to put in a voltage divider between the audio output and line in such as this:
https://www.circuitlab.com/editor/#?id=x5r3s4exfgzp

mikronwfh, 2023-08-06:
> Hi, I've created a [new branch](https://github.com/AllanGallop/mini-web-radio/tree/no-battery) with a modified firmware INO file and a schematic change (see the readme). This removes the need for [...]

I understand that there is no option to use the microUSB that we have on the ESP32 board to power the radio? I have to make a new one and connect it to 3v3 and 33?

mikronwfh, 2023-08-06:
[image]

AllanGallop, 2023-08-07:
> I just copied the new MWR.ino file and the rest remains unchanged? I have this message right now. [image]

My mistake, it should read as ``` if(digitalRead(POWER_SW) == LOW)```, I have updated the branch to reflect this. Yes the remaining structure is unchanged, if you remove the extra ```=``` from the if statement it should compile

mikronwfh, 2023-08-07:
I'll check it and let you know. Meme also another problem. After installing IDE 2.0, I had ESP32 visible immediately under COM when I connected it to USB. Now I threw out all arduino ide directories and installed v 1.8.16 again and in Tool >> Port I have inactive. Nothing happens in the device manager either when I connect the ESP32. Fix it ASAP.

AllanGallop, 2023-08-07:
I cannot comment on issues regarding the Arduino IDE or ESP libary, nor fix them as I am not a contributor or owner of those projects.

mikronwfh, 2023-08-07:
> 
Done and it works. I replaced the USB cable with another one.

mikronwfh, 2023-08-07:
[image]

Not too many exits? There is not so much on my WEMOS ESP32.
So I don't know if I'm doing it right. I understand that the microUSB on the WEMOS ESP32 board will supply power after switching it on with a potentiometer? Should I put a microUSB between the GPIO and the potentiometer?

[image]

mikronwfh, 2023-08-07:
I do not know how to connect the potentiometer with the entire ESP32 system in the new version. Instead of a battery, I have a USB cable with two wires +/- prepared. Only USB is 5V, and in the old scheme with the battery it is 3.7V. A new scheme for connecting the potentiometer with ESP32 would make it much easier for me. And should I have the power supply from the ESP32 board as it is in the old scheme, or can I use the microUSB power supply?

[image]
[image]

AllanGallop, 2023-08-08:
[image]

Hi, hopefully this clears it up for you. I've omitted the additional parts for simplicity

mikronwfh, 2023-08-08:
the potentiometer has 5 legs.  if one is bridged, there are 4 remaining. and in the diagram I have 3 connections going to the potentiometer.

AllanGallop, 2023-08-08:
You have a switched potentiometer which contains both a switch and potentiometer, the two pins on the back are the switch, the remaining 3 tabs on the side are the potentiometer; with knob facing you they are left connects to ground, center goes to pin 34, and right to 3v). you may bridge one of the switch pinsto the right (3v) leg to reduce need for multiple 3v wires with the remaining switch pin connected to pin 33

mikronwfh, 2023-08-08:
First of all, thank you very much for your commitment to running.
I just finished my project according to your idea. :) Wow! It works! :))

https://youtu.be/Dh8_dGSo-cg

One thing that puzzles me is whether the diode on the ESP32 board (LED charging bat) should be lit after turning off the power using the potentiometer? This blue diode on 5 GPIO goes out, but the factory one on the board quickly pulses with blue light.

[image]