AllanGallop/mini-web-radio #9: Is the Max fine for headphones?

olup, 2023-08-18:
This DAC is usually paired with a speaker, is that working all the same with headphones ?

AllanGallop, 2023-08-19:
It works, but using class D is not overly efficient and it's mono output so not exactly high fidelity but for the original application of project it's fine. 

I'll be releasing the plans for version 2 very soon which uses a PCM102a and a TD1308 if you prefer a headphone only design with stereo

olup, 2023-08-19:
I am building a product that will offer speaker and headphone. It's going to be mono anyway. Using just one i2s amp would be a good design. Is power consumption the only drawback then ? How bad is it ?

AllanGallop, 2023-08-19:
Sorry I should clarify, the use of a 3W class D amp solely for headphones makes a little less sense as they don't need all that much to drive them and can be more sensitive to distortion, for a small speaker it's more ideal, likewise if using both. Class D is more power efficient but lacks fidelity of say a class AB, however that doesn't really matter too much for this projects application (44.1khz mp3 stream) and the efficiency benefits battery life.

olup, 2023-08-19:
My bad, I didn't catch well. So in my case, a 44.1 mono mp3 player with speaker or optionnal headphone output, I understand the max makes actually good sense. I don't target hifi, this is mainly stories and podcasts.

Thanks for your answers.

closed 2023-08-19 by olup