AllanGallop/mini-web-radio #10: Application of PCM5102

mikronwfh, 2023-09-22:
I bought Dell ax510 speakers that are powered by 12v 1A and I wanted to ask if, after using a 12v 2 amp power supply, 

I will be able to power esp32 as well. I also want to ask about throwing away the potentiometer 

and reprogramming so that the ESP 32 board, my internet radio, always works with the maximum set value after turning it on. 
volume because I will adjust the volume from the loudspeaker. Please remember that this time I will use the pcm5102 amplifier

AllanGallop, 2023-09-25:
I don't see why not, the nominal current draw for a MWR2 which uses a PCM5201PWR and is around 100mA @ 3.3v (~330mW) so even with losses from buck conversion a 1A supply should be more than sufficient. You can programmatically specificy the output volume though the values are a bit ambiguous so I would recommend adding a trim pot to prevent clipping.

closed 2023-09-25 by AllanGallop

mikronwfh, 2023-10-06:
I'm trying to do it right now, but I don't think I know what it's supposed to look like. Can you somehow help with the code? And the diagram for PCM5102?

AllanGallop, 2023-10-06:
Functionally they are both the same as they use I²S protocol so it should be as simple as matching the signals to the correlated pins of the module. 

To maintain a volume without the potentiometer the lowest effort programmatically would be to edit line 83 of the INO with a fixed integer instead of the function call ie  ``` mRadio.setVolume(22);```

mikronwfh, 2023-10-07:
I've already managed. On PCM5102, SCK should be provided on GND. On ESP32, 12 and 14 remain free.

Now I would like to power ESP32 - 5V in some way directly from the PCB because I do not have that many cables with a USB - C plug

[image]
[image]

mikronwfh, 2023-10-07:
I'm still thinking about the potentiometer. Is it possible to change to digital? The standard one is quite large and unattractive compared to the entire design. Or maybe if not digital, then analog, only in smaller sizes. But what kind?

mikronwfh, 2023-11-12:
In my home WiFi network, I observe a case where the radio disconnects from the URL address of the radio station I am listening to.
I check the connection for my radio in the winbox of my mikrotik and it is connected to the router. What modification of the code could check whether the connection stream to the radio station's server has not been interrupted, because the connection to the router via WiFi is still functional and working.

AllanGallop, 2023-11-12:
ultimately there is not a lot that can be done as several factors are involved when it comes to drop outs on the network side such as bandwidth, contention, bitrate that do not necessarily fall under the conditions of a WiFi disconnect event.

I have observed that some of the ESP32 clones have less than stellar antenna design but even on the official expressif units it's still a comprise of size and performance.

In theory you could do something with the RSSI method but any static values would vary between boards and location. The current code will attempt to reconnect if the disconnect condition is met